avishorp/TM1637 #17: Waveforms

VendicarDecarian, 2017-02-06:


VendicarDecarian, 2017-02-06:
I audited your TM1637 bit banger library and note that on real hardware, the output waveforms on both the Clk and Data lines are triangular rather than square when the Arduino output is connected to the display chip.

When disconnected they revert to a nice square wave.

By decreasing the effective baud rate to 20,000 bps, they can be made to be reasonably square.

Further you drive down both the clock and data lines at roughly the same time.  While not invalid this can lead to issues if the waveforms are not square.

I can send images if you like.

avishorp, 2017-02-16:
This does not seem to be a software problem. Make sure you have proper pull-up resistors on your I2C bus.

closed 2017-02-16 by avishorp