avishorp/TM1637 #18: No Display

TVRleigh, 2017-02-06:
I've tried all the test codes, and I cant get any of them to work, I've used SevenSegmentTM1637 and this works fine but I would prefer to use your smaller .h .cpp 
I'm using a UNO with CLK 4, DIO 5
Anyone had this problem and found a fix.

avishorp, 2017-02-16:
What hardware module are you using?

TVRleigh, 2017-02-22:
I'm using this one from Ebay.
http://cgi.ebay.co.uk/ws/eBayISAPI.dll?ViewItemVersion&item=322148576516&view=all&tid=1708637690011

mortenfyhn, 2017-03-06:
I'm having the same issue, using a 4-digit 7-segment from Robotdyn and an NANO V3 also from Robotdyn. The SevenSegmentTM1637 library works, but I get no response with this one.

Tilleen, 2017-03-21:
From looking at the code, it looks to me like it is setting the pins to be inputs so that it will use their pullup to make the signal high, but it then turns off the pull up on the next line. It seems to be trying to use the pinmode (changing from input to output) to set the high and low rather than just setting it to be an output and the setting the output to high and low. With the current configuration it will rely on the destination pin on the TM1637 drag it high if there is no other circuitry around.

sixtemesseven, 2017-03-23:
I agree on Tillen's observation.
I wonder if the original author did it for performance reasons. I tested it on a Arduino Mega and to my surprise:

- pinMode() method takes 4-8 micros
- digitalWrite takes 8-14 micros

avishorp, 2017-03-23:
@Tilleen is correct. I2C bus relies on an external pull-up resistor to hold the line at high level while not driven. See my answer [to this issue](https://github.com/avishorp/TM1637/issues/19).
Regarding that specific hardware, I have no idea what's the problem with it. Maybe it lacks those pull-ups. Driving I2C pins high is not recommended, it might cause excessive current to flow through the pin.

Tilleen, 2017-03-23:
Just a question. Couldn't you just set the internal pull up to on, so that, that would use both the internal and external pull ups to set it high? Which may look strange. You could put in a resister in to limit the current?

For information, there is this article on digitalWrite speed: https://www.codeproject.com/articles/589299/why-is-the-digital-i-o-in-arduino-slow-and-what-ca

I have added stuff to the code to handle keyboard that the TM1637 chip has, although I have not seen a device that actually uses it.

(Don't know why my text is so large in previous post).

avishorp, 2017-05-04:
Internal pull-ups are not usually used for I2C, because they're too weak.

Tilleen, 2017-05-04:
Trying to learn here, what do you mean by too weak?

avishorp, 2017-05-04:
It means that it's resistance is too high. The value is also quite unpredictable because of the way it is implemented in the silicon.

Tilleen, 2017-05-04:
I think that is what I thought you meant. So if the resistance is high then it will be slower than would be preferred by the standard (circuit).

Thanks.

3zuli, 2018-01-08:
I also had issues with no display using Arduino UNO. I verified the signals on an oscilloscope and the problem was with signal rise time, which was relatively slow. My module has 10K pullups and some smoothing capacitors on the data lines. I fixed it by increasing the bitDelay() from 50us to 100us, this gives the signal enough time to reach a log. 1 and the display works. This is the module I am using: https://www.ebay.com/itm/0-56-TM1637-4Bit-Digital-LED-7-Segment-Clock-Tube-Display-For-Arduino-White/172971714554?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

avishorp, 2018-02-10:
Can you create a pull request with this change?

PR #30 (2018-03-20): Increases bitDelay to 100ms
Some boards on ebay have high value pullups and smoothing caps that mean that 50ms is not enough, raise that to 100ms.

resolves #18
commit 747b75ba (2018-03-20): Increases bitDelay to 100ms

Some boards on ebay have high value pullups and smoothing caps that mean that 50ms is not enough, raise that to 100ms.

resolves #18
closed 2018-03-20 by avishorp