avishorp/TM1637 #19: Incorrect use of GPIOs

msloth, 2017-02-16:
Hey, thanks for the lib. However, you are using the incorrect functions for the GPIOs: instead of using the functionality for setting pins high/low, you are setting them as input/output. This should only be done for the SDA-pin when waiting for an ACK. I think the arduino-functions for setting the output low or high is `digitalWrite()`.

msloth, 2017-02-16:
Might be the cause of #17

avishorp, 2017-02-16:
The reason for using this method is the external pull-up that exists on SDA & SCL. When the the line has to be pulled low, it is set as output. In other cases, it is set as input, allowing the external pull-up resistor to set the logic level to 'high' or any other potential bus user to pull the line to a low state.

closed 2017-02-16 by avishorp