avishorp/TM1637 #20: Central Colunm Activation

RogerMito, 2017-04-22:
Hello friends.
First of all I would like to thank you for the wonderful work you have done.
I am using a library developed by you on a personal project of a clock.
However, I am not able to activate the central column of the display. I'm using version 1.1.0 of the library. I noticed that there is no implementation for the "point" method: display.point (POINT_ON); Display.point (POINT_OFF);
can you help me?

Thank you.

avishorp, 2017-05-04:
It depends on how you use the library:

- If you're calculating the individual segment value on your own, you can use setSegments, and then  the MSB bit is usually connected to the colon or to the decimal points.
- If you're using showNumberDec, you can use the new showNumberDecEx which has control over those segments.

Sn3b, 2017-06-29:
How do you use showNumberDecEx to display the central column?

avishorp, 2017-07-09:
showNumberDecEx has an extra argument that contains a bitmask that controls the 8'th segment of each digit. This one is usually connected to the colon mark or to the decimal points. Please have a look at the test code.

ybaransky, 2018-04-02:
Is it really either the colon or the decimal point, never both? It seems the LED PCB part I am using connects only to the colon, even though the image shows the colon and the decimal points lit. Can someone post a 4-digit part hardware part that does the decimal point? The part I am using is
https://www.ebay.com/itm/0-56-inch-TM1637-4-Bits-LED-Clock-Tube-Display-for-Arduino-Red-Blue-Green-White/112674024638?hash=item1a3be52cbe:m:mU29WDo2Sfcy3aGBDVrYUNA
thx

gerrydelasel, 2018-04-06:
That display has a decimal point for each digit which can be addressed in the usual way (0bxGFEDCBA where x is the decimal point).
But the colon is separately controlled and will probably need a dedicated function.

closed 2018-10-25 by avishorp