avishorp/TM1637 #31: Do not work for me at all

morganzd, 2018-02-01:
Have this display: https://www.ebay.com/itm/0-56Inch-TM1637-4Bit-Digital-LED-Anzeige-Clock-Tube-Display-For-Arduino-Yellow/112635464786?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649
Example sketch do not work at all. Some other libraries for TM1637 worked, so I think connection between arduino and the display is ok, CLK and DIO are set correctly. Cannot find issue :(

morganzd, 2018-02-01:
Sorry, I didn't notice that same issue already opened.
Solution by 3zuli works for me. But still looking for library which would fully work with my HW, capable to display e.g. negative number with decimal point.

austinindiedev, 2018-03-27:
Would it be possible for you to share the solution 3zuli provided?

ybaransky, 2018-04-02:
TO get anything to work, I had to increase the bitDelay to 100 micros and then it worked. 50 micros is too slow. if it makes any difference, I am using a wemos d1 esp8266 to drive the display.

I am also looking to get the decimal point to work. even with all bits on in the setSegments call the dp does not light up, only the colon. I am using
https://www.ebay.com/itm/0-56-inch-TM1637-4-Bits-LED-Clock-Tube-Display-for-Arduino-Red-Blue-Green-White/112674024638?hash=item1a3be52cbe:m:mU29WDo2Sfcy3aGBDVrYUNA
and am thinking maybe its either or, colon or decimals.

avishorp, 2018-04-04:
The increase in bit time has been merged to master.

closed 2018-04-04 by avishorp