avishorp/TM1637 #32: multiple displays?

BigRedBeard86, 2018-02-22:
I am trying to find a way to use multiple tm1637 displays with different outputs. Is there a way to do this?

avishorp, 2018-03-20:
The library does not use any special HW resouces (except for GPIO  pins),
so there should be no problem hooking multiple displays. Just create a
TM1637 object instance for each display, with the appropriate CLOCK and
DATA pin numbers.

Avishay.

mcuadros, 2018-10-14:
They can share the CLOCK pins?

avishorp, 2018-10-25:
I believe there should be no problem with this.

closed 2018-10-25 by avishorp

jbellue, 2019-09-15:
I am trying to use two displays sharing a clock pin, but I can't get that to work with an Arduino uno: one of the displays shows garbage data when updated.
I have no issue using two separate pins.
I had a look through the docs, but I couldn't find anything that could cause that.
Has anybody managed to use multiple displays on only one clock pin?

Darkwing371, 2019-12-11:
Theoretically it should be possible to share the Clock pin. It could be an electrical issue in your setup. A guy in the Arduino forum seamed pretty successful with one Clock pin. Maybe it helps when you read in here:

[https://forum.arduino.cc/index.php?topic=404178.msg3823103#msg3823103](url)