avishorp/TM1637 #33: Turn Display Off?

dan2600, 2018-03-31:
I'm having trouble blanking the display. I've tried sending an empty array with setSegments but it just crashes

ybaransky, 2018-04-02:
the setBrightness() call has a parameter to enable/disable entire display. might that work for you (if it works). 
odd that all 0's in the setSegments crashes things. did you figure it out? also, have you ever gotten the decimal place to work? if yes, which LED board are you using? thx -yurij

avishorp, 2018-04-04:
Please share the code that you believe crashes the library. If you want to turn off all the segments, you have to pass an array of length 4 filled with zeros, not an empty array.

2winners, 2018-04-23:
i have added a clean() function to my version of your library saves me time coding because i found the brightness on off  to much work and anoying

closed 2018-10-25 by avishorp