avishorp/TM1637 #34: Enabling the decimal point

ybaransky, 2018-04-01:
Hi - First, thank you for writing this driver. Alot of work, little reward. 
I am using a flavor of the tm1637 IC, the ebay item is 
https://www.ebay.com/itm/112674024638?ViewItem=&item=112674024638
the LED has both the colon and the decimal points (there are number of flavors with either or, this has both). 
I can get the colon to work, but not the decimal point no matter what I have for the "dots" parameter, including all lit. 
I am using the Wemos D1 esp8266 module if that makes any difference, and I needed to use 100us for the bitDelay to get this to work at all. the 50 that you had in your library did not work.
Many Thanks
-yurij

avishorp, 2018-04-04:
Are you sure the decimal points are actually wired to the chip? As far as I understand, the chip has one additional bit per digit, so it can't control both 4 decimal points and one colon. Maybe if you figure out how the whole thing is connected I would be able to help you.

ybaransky, 2018-04-04:
It's hard to get a schematic of the actual circuit board, but I think you
are right about the hard wiring. The picture the seller has shows it
attached to the board and all decimals, colon, segments lit. I think that
is actually impossible with this board

2winners, 2018-04-23:
HEY , I have the same ebay display but i noticed mine has 12 mins and i manually checked all pins and none light up the DP, you need 14 pin display to light up the DP

closed 2018-10-25 by avishorp