avishorp/TM1637 #36: Library not recognized in Arduino IDE

Futski-III, 2018-04-13:
Arduino IDE will not install the library using install from Zip.  In addition, even after manually adding the library to the libraries folder, Arduino IDE does not recognize the library.
The Issue can be solved by adding the following line to the library.properties file:

`maintainer=`

This should be added to the file by default.

per1234, 2018-04-13:
This sort of problem is why I recommend against using PlatformIO for continuous integration of Arduino projects. If your Travis CI build had used the Arduino IDE then this issue would be caught immediately:
https://travis-ci.org/per1234/TM1637/jobs/366309423#L492

If you want to support PlatformIO then it probably makes sense to also use it in the CI builds since it's not as good at dependency resolution as the Arduino IDE. But do so in addition to the Arduino IDE, rather than instead of it.

avishorp, 2018-04-16:
Merged.

closed 2018-04-16 by avishorp