avishorp/TM1637 #41: Enable colon

blane1257, 2018-05-28:
i'm making an extremely simple clock but i cant figure out how to light the colon.
heres my code:

```
#include <TM1637Display.h>
#define CLK 4 //can be any digital pin
#define DIO 5 //can be any digital pin
TM1637Display display(CLK, DIO);

int hours = 6;      //set these to 
int minutes = 40;   //whatever time it is
bool am = false;    //define if its am or pm
const int led = 6;  //am/pm led pin

void setup() {
}

void loop() {
  display.setBrightness(1);
  display.showNumberDec(hours * 100 + minutes, false, 4, 0);
  minutes += 1;
  if (minutes >= 60)
  {
    minutes = 0;
    hours += 1;
  }
  if (hours >= 12)
  {
    hours = 1;
  }
  if (am == true)
  {
    digitalWrite(led, HIGH);
  }
  else
  {
    digitalWrite(led, LOW);
  }
  delay(60000); //60 seconds
}
```

the following code works to enable the colon:
```
uint8_t segto;
int value = 1244;
segto = 0x80 | display.encodeDigit((value / 100)%10);
display.setSegments(&segto, 1, 1);
```
but i cant figure out how to change the number being displayed

MHz000, 2018-06-30:
same problem, got:
   display.showNumberDecEx(4, 0b01000000, false, 2, 0);
  **'class TM1637Display' has no member named 'showNumberDecEx'**
any idea ow to diplay the colon?
Thank!

avishorp, 2018-07-02:
Are you using the latest version from GitHub?

MHz000, 2018-07-06:
Thank you for the hint. Using the lates version solve the problem :-))

However no dots are displayed using:
//
  display.showNumberDecEx(1234, 0x80, true, 4, 0);
  delay(TEST_DELAY);
  display.showNumberDecEx(123, 0x40, true, 4, 0);   // shows ":" after the
second digit
  delay(TEST_DELAY);
  display.showNumberDecEx(12, 0x20, true, 4, 0);
  delay(TEST_DELAY);
  display.showNumberDecEx(1, 0x10, true, 4, 0);
  delay(TEST_DELAY);
//

pls give an example how to display dots after figures
Thank you

MHz000, 2018-07-10:
shit! It looks like my TM1637 4 digit display has "blind" dots but just the
colon :-((
Any hint where a full featured display can be bought welcome

Thank-you

ybaransky, 2018-07-10:
I am also looking for such a device. Adafruit.com has a several 4-digit 7
segment displays that let you address the colon and the individual
decimals. Unfortunately it uses a different driver chip and is much more
expensive than displays that use the tm1637.

closed 2018-10-25 by avishorp