avishorp/TM1637 #47: Showing a colon

XGamerYU, 2019-01-09:
Hello folks,

so I'm building a digital clock using the ds1307 RTC module, and everything works fine, except the issue that there's no colon or a dot showing in between hours and minutes. Is there a way that I can solve this. 
Thanks in advance!

ybaransky, 2019-01-09:
If just the Colin, doable. It's in this thread somewhere.
But no one succeeded with colon and decimal. In theory yes, but looks like
the hardware does not allow it

XGamerYU, 2019-01-09:
How come that I saw numerous pictures around the internet with the same hardware, but having a dot or a colon...

jsdevel, 2019-07-31:
look at the comments here: https://github.com/avishorp/TM1637/blob/master/TM1637Display.h#L107

for a clock i do something like this:

```
display_clock.showNumberDecEx(102, 0b01000000, true, 4, 0);
```

this would display `01:02`

scottchiefbaker, 2020-05-06:
Did this ever get solved? I'd really like to have full control over each digit and the colon. Ideally I'd like to be able to do this pseudo-code:

```
setDigit(1, 1);
setDigit(2, 0);
setDigit(COLON, 1);
setDigit(3, 5);
setDigit(4, 8);
```

To have my display show "10:58"... I've used:

    display_clock.showNumberDecEx(1058, 0b11100000, true, 4, 0);

Which works, but the next iteration if I only use three digits:

    display_clock.showNumberDecEx(958, 0b11100000, true, 3, 1);

This leaves the "1" in the first position from 10:58 as this call only updates three digits. This is relevant to my real world scenario where I'm writing a countdown timer, or when 12:59 rolls to 1:00.

scottchiefbaker, 2020-05-06:
I spent a chunk of time today hacking on this issue and came up with this solution:

```
void show_clock(uint8_t hours, uint8_t minutes) {
    uint8_t d1 = hours   / 10; // Tens digit of hours
    uint8_t d2 = hours   % 10; // Ones digit of hours
    uint8_t d3 = minutes / 10; // Tens digit of minutes
    uint8_t d4 = minutes % 10; // Ones digit of minutes

    /*
    char msg[100] = "";
    snprintf(msg, 100, "Displaying: %d hours %d minutes (%d%d:%d%d)\n", hours, minutes, d1, d2, d3, d4);
    Serial.print(msg);
    */

    uint8_t digits[4];
    digits[0] = display.encodeDigit(d1);
    digits[1] = display.encodeDigit(d2); 
    digits[2] = display.encodeDigit(d3);
    digits[3] = display.encodeDigit(d4);

    // Turn on the colon
    digits[1] |= 128;

    // No leading zero on times like 1:34
    if (!d1) {
        digits[0] = 0; // Off
    }

    display.setSegments(digits, 4, 0);
}
```

On my board the colon is enabled by or'ing the second digit with 128;

MHz000, 2023-10-19:
@scottchiefbaker 
capeau! after spending houres searching the net **your code solved the problem**!
Thank you

Peanuts2000, 2024-03-25:
Hi,
> On my board the colon is enabled by or'ing the second digit with 128;

Excactly: I have a suggestion to expand the Segment bits by one:
in file "TM1637Display.h" add a line 29:
"#define SEG_P   0b10000000 // added by Kurt for access point/colon-LEDs"

This is how I access any 8th Bit for Dots or colons on 7-segment LED modules.
Regards, Kurt