avishorp/TM1637 #51: Dots and leading zero

Noisman, 2019-03-14:
It's the simple question - Why dots? Usually dots separate integral and fractional parts of a number.  Current library understand only integer numbers. For this reason we have to prepare numbers for display and convert fractional part to integer number. We have to manipulate with dots in parallel flow. And control how many fractional digits we want to display and where is the dot. 
It was an lyric introduction in trouble. A example: I want to show numbers with 2 fractional digits like that 1.25  without leading zero. And if I have to show totally zero result in the same format I'm waiting for 0.00. But library show just one zero in right position and not sowing dot.  In kind of time without leading zero we waiting  0:00 on display.
Simple way to make it possible and save backward compatibility is to convert boolean leading zeros to byte to make it possible describe how many zeros to show.

avishorp, 2019-03-19:
You are right, currently there is only a function to handle integers. If
you want to display fractional numbers, I would advise you to write it from
scratch (directly to segments) and not rely on the the existing integer
implementation.
If you do so, please make a PR with your extension.

Avishay

Noisman, 2019-03-20:
Here is my implementation for showing clock. There are eliminated sign control and different count basis due to task. I a little bit optimize number generation code based on original library.  
In this method you can set where and how many zeros you want to show. It was made in the same manner like dots coding in original library.  One zero in lowest order digit  - 0b 0001 0000, three zeros (my case) 0b 0111 0000. All 4 digits showing zerros 0b 1111 0000.

`
class myDisplay:public TM1637Display
{
	public:
		myDisplay(uint8_t pinClk, uint8_t pinDIO):TM1637Display(pinClk, pinDIO){};
		
		showDigits(unsigned int number, byte dots, byte zeros, byte length, byte position)
		{
			byte digits[length] = {0}; 
			byte digit;

			for(int i = length-1; i >= 0; i--)
			{
				digit = number % 10;

				if (number != 0 || (zeros << i) & 0x80)
				{
					digits[i]  = encodeDigit(digit);
					digits[i] |= (dots << i) & 0x80;
				}
				number /= 10;
			}
			setSegments(digits, length, position);
		}
};
`

avishorp, 2019-07-23:
Maybe it's a little more flexible, but for the sake of API stability, I will stick to the existing implementation.

closed 2019-07-23 by avishorp