avishorp/TM1637 #67: first two digits are buggy

codinghusi, 2020-06-22:
Hey, I got the module 3642BH.
In my first test the function setNumberDec having length=4 and pos=0 just works proberbly. But after I tried length=1 or length=2 and pos=0 or pos=1 the given (one digit) number will be always displayed on the first two digits repeated.
That applies also for the setSegment function.

In my observations I saw this behaviour only for the first and second display digits.

avishorp, 2020-06-27:
Can you please share the code causing the problem? I was unable to reproduce it.

codinghusi, 2020-06-28:
Yes of course, sorry :)

With this code the first two digits have activated the segments for 1 and 2 so you get a flipped 6. The third is 3 and the fourth a 4.
Probably it's a problem with my hardware because I tested another library that had the same issue.

Here is the code:
`#include <TM1637Display.h>

#define CLK 2
#define DIO 3

TM1637Display digits = TM1637Display(CLK, DIO);

void setup() {
  digits.setBrightness(7);
  digits.clear();
  digits.showNumberDec(1234, false, 4);
}

void loop() {
  
}`

avishorp, 2020-06-28:
Perferct output on my hardware.

closed 2020-07-09 by avishorp