avishorp/TM1637 #69: support for mcp23017

thijstriemstra, 2020-07-21:
Would be nice to be able to use this library with an MCP23017 expander. Something like https://github.com/adafruit/Adafruit-MCP23017-Arduino-Library and a class where I can pass in the `mcp` instance that this library would call on instead of using the global `pinMode`. One of their examples:

```cpp
Adafruit_MCP23017 mcp;
  
void setup() {  
  mcp.begin();      // use default address 0

  mcp.pinMode(0, INPUT);
  mcp.pullUp(0, HIGH);  // turn on a 100K pullup internally

  pinMode(13, OUTPUT);  // use the p13 LED as debugging
}

void loop() {
  // The LED will 'echo' the button
  digitalWrite(13, mcp.digitalRead(0));
}
```

avishorp, 2020-12-22:
Unfortunately I don't have this kind of hardware, so I'm unable to test it.

closed 2020-12-22 by avishorp

thijstriemstra, 2020-12-22:
@avishorp what if i order you an mcp23017?