avishorp/TM1637 #81: 6 digit support

wouterds, 2021-02-27:
Anyone got this working with 6 digits?

riraosan, 2021-04-11:
I think it works. Didn't it work? The four digits have moved.

wouterds, 2021-04-11:
It doesn't work properly (digits are mixed / order seems wrong), I sadly lack the knowledge to fix this myself.

riraosan, 2021-04-11:
>digits are mixed / order seems wrong

I'm guessing you bought the 6-digit LED module shown in the screenshot, but it's possible that the LED and TM1637 are misconnected in that module.
You need to confirm the facts about LED modules.
When You try to display "123456" on the LED module using the TM1637 library, what is the display order of the numbers?
ex) "456123"

BleuLlama, 2021-04-21:
I added support for these as well as a new copy of the example that uses 6 digits. I submitted a pull request for the updates. :)