avishorp/TM1637 #84: use tm1637 with pcf8574 port expander

alirezaimi, 2021-04-15:
Hi
How can i use tm1637 7segment with pcf8574 port expander ?

PCF8574 expander(0x20, 4, 5);  // esp8266 nodemcu and all Ax pins are Grounded
TM1637Display display(P4, P5);

and when i connect pins to 4 and 5 of esp8266 board its work !

avishorp, 2021-04-24:
Sorry I don't know how to use the PCF8574, but you should probably replace the pinMode and digitalWrite commands in the source code to their equivalent function for controlling the expander pins.

closed 2021-04-24 by avishorp

alirezaimi, 2021-05-03:
Thanks, but I did this and edit pcf lib code and not work .

aschewski, 2023-02-26:
@alirezaimi: I have the same challenge. Have you found a solution in the meantime to use tm1637 with pcf8574 port expander?