avishorp/TM1637 #88: ESP32 tuning

thenicnic, 2021-06-07:
Great library, thanks!

If you have problems with your TM1637 display (in my case a 4-digit 7-segment):

- [Paul_B gave the clever tip](https://forum.arduino.cc/t/tm1637-not-working-with-esp8266/602671/4) to remove the two 100 nF SMD capacitors - this is really neccessary since they should be in pF range and cause trouble than solving problems when they have a much too high capacity on higher clock rates like ESP32's up to 240 MHz. (My exemplaries also had two 200 nF caps - no need to remove these.)
[image]

- My displays caused WiFi loss occasionally when the displays were powered by the same 5 V voltage source like the ESP, especially at startup/reset. I fixed this by adding a N14007 diode to the 5 V input of the LED display.

RalphBacon, 2021-10-04:
@thenicnic Thanks for the suggestion on this.

However, I've managed to get this working today, no removal of capacitors (yet, I'll see how it goes) but I was intrigued that you were running this display **from 5v**. This means both CLK and DIO are at 5v, not compatible with the EP32s 3v3 maximum on any pin.

I run mine from the TTGO v1.7.1 on-board 3v3 pin (ironically this is at 3v8 for reasons unknown, I've raised an issue with TTGO) and it's all working fine so far.

Any comment on your use of 5v? Are you using level shifters or a simple resistor potential divider? Or just plain 5v?

thenicnic, 2021-10-04:
> Any comment on your use of 5v? Are you using level shifters or a simple resistor potential divider? Or just plain 5v?

I should have mentioned that I used an ESP32 "Dev Board" with a voltage regulator on it. It's capable of handling 5 to 12 V DC (but higher voltages are not so recommended because of the heat).

Your 3,8 V board is indeed strange. Did you measure under load? And what's the input voltage? Maybe a not-so-great regulator.

RalphBacon, 2021-10-05:
@thenicnic 
Yes, my board has an on-board 5v LDO too, but if I power the TM1637 with that 5v, that 5v voltage appears on DIO and CLK of the TM1637 (according to my multimeter). And even though those pins are INPUT pins to the TM1637, we would still present the ESP32 with 5v on only 3.3v tolerant pins. Is this not the case? As you don't seem to have had problems I guess you will be fine continuing to use 5v on the TM1637's VCC pin. In my case it seems to work just fine on 3.3v (3.8v!) with no risk to the ESP32. 

Regarding the 3.8v out of my "3.3v" regulator, I checked the schematic against what I have. It is definitely a genuine TTGO T8 but that voltage regulator seems very bad. It's powered from USB 5v (barely, more like 4.8v). Unfortunately it's a 5 pin regulator device and I only have 3 pin devices. I still might be able to replace it. I'm certainly not happy using this particular module in the field. I'm astounding the ESP32 is not releasing the magic smoke at the absolute maximum voltage (stressed) is 3.6v - strange!

Thanks for your reply, it's all very interesting what the ESP32 will withstand.

thenicnic, 2021-11-09:
@RalphBacon Thank you really much for investigating and your good work at finding the cause. You're absolutely right - mine does also communicate with 5 V which is a flaw!
The diode I added dropped the voltage by 0,7 V so I assumed that the logic voltage does get low enough to get into the sampling range of the ESP32's DIOs. This would have been an explanation for the WiFi drops. 
So I completely rearranged my setup for 3,3 V (including setting the DC-DC to 3,3 V and bypassing the ESP32's dev board regulator) and hoped that this will fix the WiFi issue without using a diode.
But bummer! I got no display on the TM1637 LED display unit and the WiFi issues still occur! Even if I add the diode (although I think that 2,6 V are too less for the LED display unit) the WiFi dropout at boot keeps happening.
Don't know what to do. I think I will buy some higher quality LED displays.

thenicnic, 2021-11-09:
Oh no my fault! I was using reserved GPIOs on my rebuild. I remapped the LED display to pins 26 and 27 and everything is working fine on 3,3 V! Display is a bit darker but no problems at all although no diode is in the circuit.

RalphBacon, 2021-11-10:
> Oh no my fault! I was using reserved GPIOs on my rebuild. I remapped the LED display to pins 26 and 27 and everything is working fine on 3,3 V! Display is a bit darker but no problems at all although no diode is in the circuit.

The absolute solution would be to use level shifters on the pins 26 & 27 then you could use 5v on the display and make it nice and bright! 

In this case I think a simple two-resistor divider (2k2 from the display to the ESP pin, and a 3k3 from the ESP pin to GND would drop 5v to around 3v - I do this, but do check with a multimeter). So, four resistors in total for your project would be sufficient.

But I am glad you found the cause of the issue and now have it working.

thenicnic, 2021-11-10:
Yes lowering the logic level is the optimal solution but since I changed the circuit now to run at 3,3 V I won't rework it again because I'm fine with the brightness. I set it to 15 (0x0F) which is maximum and even with a dark tint foil (for better contrast) I cannot complain. 
Also the device is primary used with web interface so display is not that important.

[image]

Thank you again!

RalphBacon, 2021-11-11:
Indeed, your photo shows a good brightness from the LED display and the foil certainly increases the contrast nicely. I'm happy you got it all working.

DrJaymz, 2022-09-12:
TLDR - will be fine on a 3v3 MCU like esp32 / 8266 etc.

From the datasheet, the IO lines are open collector.  Most boards implement a 10k pullups to 5V so if you're using a 3v3 MCU then these will try to pull higher than 3v3.  

All MCU (not CPU's) silicon have supply rail clamp diodes that clamp inputs to the 3v3 rail.  With that clamped to 3v3 the 10k pullup resistors will be seeing 1v7 which means they will limit the diode clamp current to 0.2mA which is well within their capability of a few mA and in fact they have an easier time than the output transistors trying to pull down from 5V.