avishorp/TM1637 #94: Add diagram to documentation that shows which segment is which (for SEG_A, SEG_B, etc.)

DeflateAwning, 2022-12-21:
For doing something like [this "dOnE" example](https://github.com/avishorp/TM1637/blob/3cca19607013c49f6708b7f492765fea835431ca/examples/TM1637Test/TM1637Test.ino#L11), it would be helpful to have a diagram which shows which segment is which.

DeflateAwning, 2022-12-21:
[image]

It appears that this is an example of the diagram I'm thinking of

Ardjoen575, 2023-02-26:
sketch_feb27b.ino:1:20: fatal error: TM1637.h: No such file or directory

Ardjoen575, 2023-02-26:
#include <TM1637.h>
#include <EEPROM.h>
TM1637 display(2, 3); // Data pin and clock pin
const int INCREMENT_PIN = 4;
const int DECREMENT_PIN = 5;
const int ENTER_PIN = 6;
const int COIN_OK_PIN = 7;
const int WAIT_PIN = 8;
const int MOTOR_START_PIN = 9;

int currentValue = 0;
int previousValue = 0;
int targetValue = 0;
int coinValue = 0;
int eepromAddress = 1000;

void setup() {
  Serial.begin(9600);

  display.init(); // Initialize the display
  display.setBrightness(0x0f); // Set the brightness to maximum
  display.showNumberDec(currentValue); // Display the initial value

  pinMode(INCREMENT_PIN, INPUT);
  pinMode(DECREMENT_PIN, INPUT);
  pinMode(ENTER_PIN, INPUT);
  pinMode(COIN_OK_PIN, INPUT);
  pinMode(WAIT_PIN, INPUT);
  pinMode(MOTOR_START_PIN, OUTPUT);
  digitalWrite(MOTOR_START_PIN, LOW);
}

void loop() {
  int increment = digitalRead(INCREMENT_PIN);
  int decrement = digitalRead(DECREMENT_PIN);
  int enter = digitalRead(ENTER_PIN);
  int coinOk = digitalRead(COIN_OK_PIN);
  int wait = digitalRead(WAIT_PIN);

  if (increment == HIGH) {
    currentValue++;
    display.showNumberDec(currentValue);
    delay(300);
  }

  if (decrement == HIGH) {
    currentValue--;
    if (currentValue < 0) {
      currentValue = 0;
    }
    display.showNumberDec(currentValue);
    delay(300);
  }

  if (enter == HIGH) {
    EEPROM.write(eepromAddress, currentValue);
    eepromAddress++;
    if (eepromAddress == 1024) {
      eepromAddress = 0;
    }
  }

  if (coinOk == LOW) {
    digitalWrite(MOTOR_START_PIN, HIGH);
    delay(100);
    targetValue = EEPROM.read(eepromAddress);
    coinValue = 0;

    while (coinValue < targetValue) {
      display.showNumberDec(targetValue - coinValue);
      delay(90);

      coinOk = digitalRead(COIN_OK_PIN);
      if (coinOk == LOW) {
        coinValue += EEPROM.read(eepromAddress);
      }

      wait = digitalRead(WAIT_PIN);
      if (wait == HIGH) {
        digitalWrite(MOTOR_START_PIN, LOW);
        delay(12000);
        digitalWrite(MOTOR_START_PIN, HIGH);
      }

      delay(90);
    }

    digitalWrite(MOTOR_START_PIN, LOW);
    delay(1000);
  }

  display.showNumberDec(0); // Clear the display
}

Ardjoen575, 2023-02-26:
Plz edit a code as per your library or modify library thank you

DeflateAwning, 2023-02-27:
Your issue is that you don't have the library setup in the Arduino IDE.

Please delete your comments. This is the wrong place to ask for basic Arduino help. Please make a new issue if you still need help.